moonlight-stream/moonlight-embedded #820: Support for OSMC in moonlight-embedded/setup.deb.sh

kurac338, 2021-07-26:
Moonlight Embedded version: 2.5.0
Moonlight Embedded running on: Raspberry Pi
Moonlight Embedded running on distribution: OSMC

Verbose output `-verbose` of Moonlight Embedded:

`NOPE: Checking if upstream install config is OK ...
 ^^^^: ... Failed to fetch configuration for your OS distribution release/version.
 >>>>:
 >>>>: It looks like we don't currently support your distribution release and
 >>>>: version. This is something that we can fix by adding it to our list of
 >>>>: supported versions (see contact us below), or you can manually override
 >>>>: the values below to an equivalent distribution that we do support:
 >>>>:
 >>>>: ... distro=osmc  version=2020.11-1  codename=November  arch=armv7l  
 >>>>:
 >>>>: You can force this script to use a particular value by specifying distro,
 >>>>: version, or codename via environment variable. E.g., to specify a distro
 >>>>: such as Ubuntu/Xenial (16.04), use the following:
 >>>>:
 >>>>:   <curl command> | distro=ubuntu version=16.04 codename=xenial sudo -E bash`

What is the expected result?
Installation script recognizes the distro and proceeds installing the package.

What happens instead of that?
See provided console log output.

cgutman, 2021-07-26:
Hmm, that's not good. I assume OSMC is compatible with Raspbian Buster packages?

Can you try this command instead?
```
curl -1sLf 'https://dl.cloudsmith.io/public/moonlight-game-streaming/moonlight-embedded/setup.deb.sh' | distro=raspbian codename=buster sudo -E bash
```

kurac338, 2021-07-26:
> Hmm, that's not good. I assume OSMC is compatible with Raspbian Buster packages?
> 
> Can you try this command instead?
> 
> ```
> curl -1sLf 'https://dl.cloudsmith.io/public/moonlight-game-streaming/moonlight-embedded/setup.deb.sh' | distro=raspbian codename=buster sudo -E bash
> ```

Yep that worked. Just noting the issue for novice users that might not know how to prepare the override arguments (distro,codename,version,...).

cgutman, 2021-07-26:
Thanks for confirming. I updated the instructions on the wiki to specify the `distro` and `codename` explicitly.

kurac338, 2021-07-27:
Yesterday I have successfully updated from the compiled source v2.4.8 to distributable package v2.5.0.
Everything is working fine, or even better than before, thanks to @cgutman.

If you are updating from the old repository and package maintainer (itimmer) you might want to do the following, in order to clean the redundant files:
```bash
# Remove old itimmer.gpg repository key
sudo apt-key del 66EB14EE
sudo apt-get purge moonlight-embedded
sudo apt autoremove
sudo apt clean
```

If you need to remove a compiled package (example for OSMC distro) use:
```bash
# Remove deployed binaries, libraries and package manual
sudo rm -rf /usr/local/lib/libmoonlight* /usr/local/bin/moonlight /usr/local/share/moonlight/ /usr/local/etc/moonlight.conf /usr/local/lib/libgamestream* /usr/local/lib/libmoonlight* /usr/local/share/man/man1/moonlight.1
```

Afterwards its as easy as following the Wiki guide: https://github.com/moonlight-stream/moonlight-embedded/wiki/Packages

closed 2021-07-27 by kurac338