moonlight-stream/moonlight-embedded #823: Arcsde controller left stick sends triggers

phi-alek, 2021-08-07:
Geforce experience: 3.23.0.74
Moonlight embedded Version: 2.5
Moonlight source: this github via retropie
Pi 4 4gb
Power Supply: apple macbookpro m1 power supply
RetroPie Version: 4.7.16
Built From: Pre made SD Image on RetroPie website
USB Devices connected: 2 x arcade sticks with 16 buttons (8 for each player) (dragon rise controller), 2 usb keboard
Bluetooth devices: 2 x ipega gamepad

I used moonlight a lot in my raspberry pi 4 arcade cabinet with arcade sticks and buttons (dragon rise controller). Everything worked perfectly. But since a couple of days my dragon rise arcade sticks don't work properly, maybe it had to do with an update I did.

Everything is still mapped right (mapping wasnt necessary), but now the arcade stick controls the left and right triggers as well. I have the joystick working as the left analog stick, but now it sends left and right triggers as well. When i turn the joystick to the left, there are no trigger signals, when its in the middle then its sending half triggers signals (like analog half pressed), when turned to the right it sends full trigger signals.

I have tried everything. Deinstalled and cleaned moonlight, updated it, reinstalled it. I have even made custom mapping which was a pain because i couldn't skip buttons.
Nothing solves this problem. By the way i have 2 ipega controller connected as well wich work normally also steam link behaves normally as well.
Is there anything i can do?

cgutman, 2021-08-07:
Can you post the log from moonlight so we can tell what mapping it was using for your arcade sticks?

phi-alek, 2021-08-07:
Thank you so much for the reply.

Where can i find those? I was searching for hours before.

tomblind, 2021-08-07:
This is similar to the issues I was having. The PR I put up may fix it: #824

phi-alek, 2021-08-08:
Wow, I will try that as soon as i can. I am sitting for about 30 hours trying everything i can, to get my cabinet working again. On the other hand i have learned a bit in the process. I hope that will be the solution.

cgutman, 2021-08-08:
The fix is checked into the master branch, so you should be able to just have RetroPie update your package from source to pick up the change.

phi-alek, 2021-08-08:
Its working again. Im so happy thank you.

closed 2021-08-08 by phi-alek