moonlight-stream/moonlight-embedded #825: Unpair command not functioning

clarkemw, 2021-08-19:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.23.0.74
**Moonlight Embedded version:** 2.5.1
**Moonlight Embedded running on:** Raspberry Pi
**Moonlight Embedded running on distribution:** Raspbian

**Verbose output `-verbose` of Moonlight Embedded:** 
```
>moonlight unpair
Moonlight Embedded 2.5.1 (PI;MMAL;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.0.205...
GPU: NVIDIA GeForce RTX 3080, GFE: 3.23.0.74 (gs_04_31_29971906, 7.1.431.0)
Succesfully unpaired

>moonlight pair
Moonlight Embedded 2.5.1 (PI;MMAL;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.0.205...
GPU: NVIDIA GeForce RTX 3080, GFE: 3.23.0.74 (gs_04_31_29971906, 7.1.431.0)
Please enter the following PIN on the target PC: 6888
Failed to pair to server: Already paired
```
**What is the expected result?**
After an unpair operation I should be prompted to enter the authentication code again if I run the pair command.
**What happens instead of that?**
The unpair command states that it was successful, but if I try to run pair after running unpair gives me the following message: "Failed to pair to server: Already paired" which seems to indicate that the unpair command didn't do anything.

cgutman, 2021-08-21:
This isn't actually a Moonlight regression. The unpair command simply doesn't work on modern versions of GFE.

I don't think the official clients have used it for a long time (if they ever did). The only place I've seen it used now is during pairing to terminate a pairing attempt due to a validation error.

It makes sense that NVIDIA has removed it, because unpairing provides no benefit. There's no paired PC limit, so there's no need to unpair before uninstalling Moonlight or anything like that. It doesn't provide a useful security boundary because it requires you trust the very client you are unpairing (which presumably you're unpairing because it's untrustworthy). A malicious client could easily be patched to fake like it's unpaired but not actually unpair, thus making it useless as a security feature. The only safe way to unpair a client is on the host-side because it doesn't require trusting the client.

clarkemw, 2021-08-21:
Thank you for the background. Might be worth adding a comment in the Wiki that states that unpair is no longer supported by GFE (or just remove the description of that command option altogether)

closed 2021-08-21 by clarkemw