moonlight-stream/moonlight-embedded #828: stream not start

jibtga, 2021-08-25:
Please provide the following info.

**_NVidia Geforce Experience version:_**
GFE: 3.23.0.74

**Moonlight Embedded version:**
moonlight-enbedded_2.5.0-1buster1_armhf.deb 

**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
rasberry PI 3
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
OSMC (buster distro)

**Verbose output `-verbose` of Moonlight Embedded:**
Moonlight Embedded 2.5.0 (PI;MMAL;ALSA;EMBEDDED)
Connect to 172.18.2.103...
GPU: NVIDIA GeForce GTX 1070, GFE: 3.23.0.74 (gs_04_31_29971906, 7.1.431.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Detected Wireless Controller (050000004c050000cc09000000810000) on /dev/input/event2 as PS4 Controller
Detected Wireless Controller Touchpad (050000004c050000cc09000000810000) on /dev/input/event0 as PS4 Controller
Not mapping Wireless Controller Touchpad as a gamepad
Detected Wireless Controller Motion Sensors (050000004c050000cc09000000810000) on /dev/input/event1 as PS4 Controller
Ignoring accelerometer: Wireless Controller Motion Sensors
Stream 1920 x 1080, 60 fps, 20000 kbps
RTSP port: 48010
Initializing platform...done
Resolving host name...done
Initializing audio stream...done
Starting RTSP handshake...Audio port: 48000
Video port: 47998
Control port: 47999
done
Initializing control stream...done
Initializing video stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Received first audio packet after 0 ms
Initial audio resync period: 650 milliseconds
Received first video packet after 0 ms
Assigned Wireless Controller as player 1
Unable to recover audio data block 1456 to 1459 (2+0=2 received < 4 needed)
Network dropped audio data (expected 1458, but received 1460)
Unable to recover audio data block 1488 to 1491 (3+0=3 received < 4 needed)
Network dropped audio data (expected 1491, but received 1492)
Unable to recover audio data block 1508 to 1511 (1+0=1 received < 4 needed)
Network dropped audio data (expected 1509, but received 1512)
Unable to recover audio data block 1536 to 1539 (1+0=1 received < 4 needed)
Network dropped audio data (expected 1537, but received 1540)
Unable to recover audio data block 1552 to 1555 (4+0=4 received < 4 needed)
......................................................................
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed: -1
Network dropped audio data (expected 3109, but received 3112)
done
Stopping video stream...done
Stopping control stream...ENet peer acknowledged disconnection
done
Cleaning up input stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up audio stream...done
Cleaning up platform...done
Sending app quit request ...

**What is the expected result?**
Stream APP to TV/osmc

**What happens instead of that?**
The app is launched in PC but in TV/osmc show verbose logs and a buzz sound

fabien-github, 2022-01-07:
Check this issue:
https://github.com/moonlight-stream/moonlight-embedded/issues/776

jibtga, 2022-01-19:
The system(kodi-OSMC) not work properly when  I change from dtoverlay=vc4-kms-v3d,cma-512 to dtoverlay=vc4-fkms-v3d,cma-512
Thanks

cgutman, 2022-01-19:
@fabien-github if you need support for using `vc4-kms-v3d` (Full KMS), you can use Moonlight Qt. https://github.com/moonlight-stream/moonlight-docs/wiki/Installing-Moonlight-Qt-on-Raspberry-Pi-4

It officially supports HEVC (but not H.264) with the Full KMS driver. You can get H.264 to work though if you need by launching it with `H264_DECODER_HINT=h264_v4l2m2m moonlight-qt`.

jibtga, 2022-01-20:
Thanks, I try it

jibtga, 2022-01-20:
Yes, the CEC doesn't work, thanks

fabien-github, 2022-01-21:
Hello @cgutman ! I tried on my setup but it didn't worked for me.

I'm still running on a Pi3 / 32bit osmc, so i'm not sure if your workaround can apply here. Starting from the console, I got these kind of errors during decoding, with a black screen :
```
00:00:26 - SDL Info (0): IDR frame request sent
00:00:27 - SDL Info (0): Video decode unit queue overflow
00:00:27 - SDL Info (0): IDR frame request sent
00:00:27 - FFmpeg: [h264_v4l2m2m @ 0x2ee6810] Unexpectedly large retry count: 25600:00:27 - SDL Warn (0): avcodec_send_packet() failed: Input/output error
00:00:27 - SDL Info (0): Requesting IDR frame on behalf of DR
00:00:27 - SDL Info (0): IDR frame request sent
00:00:27 - SDL Info (0): Video decode unit queue overflow
00:00:27 - SDL Info (0): IDR frame request sent
```

If i switch back to `vc4-fkms-v3d` with `h264_mmal`, moonlight-qt works fine.

Don't know if you want to investigate this, i will probably switch to a Pi4 in the futur anyway.

Thank for your work on moonlight, it's really appreciated !