moonlight-stream/moonlight-embedded #832: Moonlight return Illegal Instruction and nothing else

LordBoBCUP, 2021-09-06:
Please provide the following info.

_NVidia Geforce Experience version:3.23
Moonlight Embedded version: 2.5.1-1 armhf
Moonlight Embedded source: repository - from steps in wiki
Moonlight Embedded running on: Raspberry Pi
Moonlight Embedded running on distribution: Raspbian

Verbose output `-verbose` of Moonlight Embedded: Illegal Instruction (literally this is it)

What is the expected result? Moonlight working as it does on my other raspberry pi

What happens instead of that? CLI returns `Illegal Instruction` and exists. No matter what combination of arguments supplied

No matter what command is issued (moonlight peer 10.221.73.15) etc will always return illegal instruction, no other output even if --debug is supplied. Has anyone else seen this before or have any further troubleshooting steps to try?

mariotaku, 2021-10-06:
What is the model version of your Pi and what is the version of Raspbian? Illegal Instruction usually caused by binary incompatibility with OS/CPU.

electrophoenix, 2021-11-09:
This is happening to me on a raspberry pi zero w after following the instructions for installing the raspberry pi package in the wiki. Fresh buster install; the script added the repo; moonlight installed without issues.

electrophoenix, 2021-11-11:
My issue is resolved after building from source

closed 2021-12-19 by cgutman