moonlight-stream/moonlight-embedded #833: New Raspbian Bullseye release, fails to install

WilliUreta, 2021-11-11:
Please provide the following info.

**_NVidia Geforce Experience version:_**
None

**Moonlight Embedded version:**
Latest commit: 81322a2b91aaaa18bdbb0be01b2c245ed000abe4

**Moonlight Embedded source:** from github, using https://github.com/moonlight-stream/moonlight-embedded/wiki/Full-details-for-Compilation-and-Installation-in-Raspbian-Or-OSMC

**Moonlight Embedded running on:** Raspberry Pi 3B+
**Moonlight Embedded running on distribution:** Raspbian Bullseye

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?**
Installation successful 
**What happens instead of that?**
Installation/compilation fails. Lots of modules not found, such as: libevdev, libudev, libva...
The new release has made some changes on the video drivers for the RPi.
https://www.raspberrypi.com/news/raspberry-pi-os-debian-bullseye/

[CMakeError.log](https://github.com/moonlight-stream/moonlight-embedded/files/7520457/CMakeError.log
[CMakeOutput.log](https://github.com/moonlight-stream/moonlight-embedded/files/7520459/CMakeOutput.log)
)

Cageman1984, 2021-11-21:
I tried fixing it with adding the old buster repository tot sources.list. installation is succesfull, but I run into a new error (and haven't looked into that yet)

But you could try the same: add buster to sources.list

WilliUreta, 2021-11-22:
I understand that there are workarounds for most of the issues, but it would be neat if we could get a Bullseye specific update (maybe not even RPi specific). AFAIK, the RPi announcement mentions stuff about the KMS video driver that may affect ME. 
Thanks anyways!

TimCurvature, 2021-11-25:
Hey there ! I am also having troubles installing on Bullseye. The installation can't install libcec4. Can someone tell me which version of raspbian i can use to get moonlight working? Steamlink App failed starting as well sadly, so i guess there's no other option but to switch to another os version and do it all over again ^^ . Would be absolutely great if we could have a List in the Readme where you can just see what Versions are known to be working and which not.

cgutman, 2021-12-19:
This is fixed by https://github.com/moonlight-stream/moonlight-embedded/commit/ba8a23725d941732eda5a9b52c40817331640c4b

Planning to get Bullseye packages published soon.

closed 2021-12-19 by cgutman

cgutman, 2021-12-20:
The v2.5.2 release is out with Buster and Bullseye packages.