moonlight-stream/moonlight-embedded #836: Raspberry Pi4B hardware decoding

liujiayu2, 2021-12-23:
My hardware is Raspberry Pi 4B based on a 64-bit official system. By following this step https://github.com/moonlight-stream/moonlight-embedded/wiki/Compilation, I build the Moonlight-Stream. But in this process, I have got a notice (missing: OPENMAX) to prompt me that can not find Broadcom, which means I can not build the hardware decoding lib (libmoonlight-pi.so or libmoonlight-mmal.so)， resulting in the video lag.
 However, when I follow your method to install  Moonlight-Stream, it runs well on my Raspberry Pi this time ( It can find hardware decoding lib libmoonlight-pi.so ). 
These steps like follow:
1. curl -1sLf 'https://dl.cloudsmith.io/public/moonlight-game-streaming/moonlight-embedded/setup.deb.sh' | distro=raspbian codename=buster sudo -E bash
2. sudo apt install moonlight-embedded

 Could you kindly tell me why your method can solve the problem above and how did you find this method? Thanks a lot.

cgutman, 2021-12-23:
64-bit Raspberry Pi OS is not supported, because it does not support MMAL decoding APIs. MMAL is the only API that can currently handle 1080p 60 FPS H.264 content on the Pi 4 with acceptable latency.

liujiayu2, 2021-12-23:
> 64-bit Raspberry Pi OS is not supported, because it does not support MMAL decoding APIs. MMAL is the only API that can currently handle 1080p 60 FPS H.264 content on the Pi 4 with acceptable latency.

lot of thanks, how can i build the 32-bit moonlight-embedded？  how you build the application on the apt source,  i want build the same 32-bit application like the "sudo apt install moonlight-embedded" myself ,lots of thanks.

cgutman, 2021-12-23:
If you're asking about building/running a 32-bit version of Moonlight Embedded on 64-bit Raspberry Pi OS, I don't think that's possible either. Even if you installed the MMAL libraries from a 32-bit installation of the OS, I don't think they will work.

I recommend downloading the regular Raspberry Pi OS which is 32-bit rather than trying to make it work on 64-bit.

liujiayu2, 2021-12-23:
> If you're asking about building/running a 32-bit version of Moonlight Embedded on 64-bit Raspberry Pi OS, I don't think that's possible either. Even if you installed the MMAL libraries from a 32-bit [...]

ok,got it ,lots of thanks

liujiayu2, 2021-12-23:
one more question：i build  Moonlight Embedded on 32-bit Raspberry Pi OS successfully, how to make it to be a package?
so that i can move the package run on 64-bit Raspberry Pi OS. i find your package (use apt get ) can run on 64-bit Raspberry Pi OS successfully.

cgutman, 2021-12-23:
I package Moonlight Embedded using a Docker container.

You can build packages via the `./build-rpi-buster.sh` and `./build-rpi-bullseye.sh` scripts in https://github.com/cgutman/moonlight-embedded-packaging

You will need Docker and some way of running armhf code. If you're running it on the Pi itself with a 32-bit OS, that should work. Otherwise, you'll need to install QEMU user emulation packages `qemu binfmt-support qemu-user-static` (assuming a Debian/Ubuntu-based distro)

liujiayu2, 2021-12-23:
ok，lots of thanks @

closed 2021-12-23 by liujiayu2