moonlight-stream/moonlight-embedded #837: Platform 'auto' not found

espenol, 2022-01-01:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.24.0.126
**Moonlight Embedded version:** 2.5.2
**Moonlight Embedded source:** Not sure what this means? Installed via the Bullseye command found [here](https://github.com/moonlight-stream/moonlight-embedded/wiki/Packages#additional-required-setup-steps-for-raspbian-bullseye)
**Moonlight Embedded running on:** Raspberry Pi 4 2GB
**Moonlight Embedded running on distribution:** Raspberry Pi OS, headless, Bullseye

**Verbose output `-verbose` of Moonlight Embedded:**
`pi@raspberrypi:/ $ moonlight stream
Searching for server...
Connect to 192.168.1.165...
Platform 'auto' not found`

If I try to specify platform using `-platform pi` I get `Platform 'pi' not found`

**What is the expected result?**
Establish connection to my desktop PC and start streaming to Raspberry Pi connected to TV

**What happens instead of that?**
Nothing, only given the very short error message above.

cgutman, 2022-01-02:
On Bullseye, you should use `-platform mmal` not `-platform pi`, though the default `auto` should try both. There must be some issue preventing MMAL from working on your device.

Did you do anything else to your `/boot/config.txt` other than the steps there in the wiki?

espenol, 2022-01-24:
Hello again, I repeated the steps in the wiki and now it seems to work properly. Not sure what went wrong. Thanks a lot for trying to help!

closed 2022-01-24 by espenol