moonlight-stream/moonlight-embedded #838: Can't compile on Raspberry Model B Revision 2.0

esistgut, 2022-01-05:
I'm trying to compile moonlight-embedded on this ancient piece of history, I'm not really sure if it is expected to compile or run so feel free to close this issue with a "LOL" stamp.
I first tested the prebuilt binary but it returns just an `Illegal Instruction` message, so following this issue https://github.com/moonlight-stream/moonlight-embedded/issues/832 I tried to compile it from source.

The base OS is Raspberry OS Lite, `Linux raspberrypi 5.10.63+ #1488 Thu Nov 18 16:14:04 GMT 2021 armv6l GNU/Linux`

This is my output:
```
pi@raspberrypi:~ $ sudo apt install git libopus0 libexpat1 libasound2 libudev1 libavahi-client3 libcurl4 libevdev2 libssl-dev libopus-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libevdev-dev libexpat1-dev libpulse-dev uuid-dev cmake gcc g++
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
cmake is already the newest version (3.18.4-2+rpi1+deb11u1).
g++ is already the newest version (4:10.2.1-1+rpi1).
gcc is already the newest version (4:10.2.1-1+rpi1).
git is already the newest version (1:2.30.2-1).
libavahi-client-dev is already the newest version (0.8-5).
libavahi-client3 is already the newest version (0.8-5).
libcurl4 is already the newest version (7.74.0-1.3+deb11u1).
libcurl4-openssl-dev is already the newest version (7.74.0-1.3+deb11u1).
libevdev-dev is already the newest version (1.11.0+dfsg-1+rpi1).
libevdev2 is already the newest version (1.11.0+dfsg-1+rpi1).
libexpat1 is already the newest version (2.2.10-2).
libexpat1-dev is already the newest version (2.2.10-2).
libopus-dev is already the newest version (1.3.1-0.1).
libopus0 is already the newest version (1.3.1-0.1).
libssl-dev is already the newest version (1.1.1k-1+deb11u1).
libudev-dev is already the newest version (247.3-6+rpi1).
libudev1 is already the newest version (247.3-6+rpi1).
uuid-dev is already the newest version (2.36.1-8).
libasound2 is already the newest version (1.2.4-1.1+rpt2).
libasound2-dev is already the newest version (1.2.4-1.1+rpt2).
libpulse-dev is already the newest version (14.2-2+rpi1).
The following package was automatically installed and is no longer required:
  alsa-base
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

pi@raspberrypi:~ $ sudo apt install libraspberrypi-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libraspberrypi-dev is already the newest version (1:2+git20211125~155417+14b90ff-2).
The following package was automatically installed and is no longer required:
  alsa-base
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

pi@raspberrypi:~ $ git clone https://github.com/moonlight-stream/moonlight-embedded.git
Cloning into 'moonlight-embedded'...
remote: Enumerating objects: 6399, done.
remote: Counting objects: 100% (359/359), done.
remote: Compressing objects: 100% (231/231), done.
remote: Total 6399 (delta 233), reused 230 (delta 127), pack-reused 6040
Receiving objects: 100% (6399/6399), 10.28 MiB | 889.00 KiB/s, done.
Resolving deltas: 100% (3846/3846), done.

pi@raspberrypi:~ $ cd moonlight-embedded

pi@raspberrypi:~/moonlight-embedded $ git submodule update --init --recursive
Submodule 'third_party/SDL_GameControllerDB' (https://github.com/gabomdq/SDL_GameControllerDB.git) registered for path 'third_party/SDL_GameControllerDB'
Submodule 'common' (https://github.com/moonlight-stream/moonlight-common-c.git) registered for path 'third_party/moonlight-common-c'
Cloning into '/home/pi/moonlight-embedded/third_party/SDL_GameControllerDB'...
Cloning into '/home/pi/moonlight-embedded/third_party/moonlight-common-c'...
Submodule path 'third_party/SDL_GameControllerDB': checked out 'aecb6b0df26c20f83868111ae4c8b66649c59e69'
Submodule path 'third_party/moonlight-common-c': checked out '8abc371fb4c970b7cfae0860789e98fbb90ed231'
Submodule 'enet' (https://github.com/cgutman/enet.git) registered for path 'third_party/moonlight-common-c/enet'
Cloning into '/home/pi/moonlight-embedded/third_party/moonlight-common-c/enet'...
Submodule path 'third_party/moonlight-common-c/enet': checked out '8d69c5abe4b699e7077395e01927bd102b3ba597'

pi@raspberrypi:~/moonlight-embedded $ mkdir build
[... 119 lines cut ...]
[ 63%] Building C object libgamestream/CMakeFiles/gamestream.dir/__/third_party/h264bitstream/h264_sei.c.o
[ 65%] Building C object libgamestream/CMakeFiles/gamestream.dir/__/third_party/h264bitstream/h264_stream.c.o
[ 66%] Linking C shared library libgamestream.so
[ 66%] Built target gamestream
Scanning dependencies of target moonlight-mmal
[ 68%] Building C object CMakeFiles/moonlight-mmal.dir/src/video/mmal.c.o
[ 69%] Linking C shared library libmoonlight-mmal.so
[ 69%] Built target moonlight-mmal
Scanning dependencies of target moonlight-pi
[ 71%] Building C object CMakeFiles/moonlight-pi.dir/src/video/pi.c.o
In file included from /home/pi/moonlight-embedded/src/video/pi.c:41:
/home/pi/moonlight-embedded/./third_party/ilclient/ilclient.h:46:10: fatal error: IL/OMX_Broadcom.h: No such file or directory
   46 | #include "IL/OMX_Broadcom.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/moonlight-pi.dir/build.make:82: CMakeFiles/moonlight-pi.dir/src/video/pi.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:171: CMakeFiles/moonlight-pi.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```

The IL stuff appears to be on my system:
```
pi@raspberrypi:~/moonlight-embedded/build $ ls /opt/vc/include/IL/
OMX_Audio.h  OMX_Broadcom.h  OMX_Component.h  OMX_Core.h  OMX_ILCS.h  OMX_Image.h  OMX_Index.h  OMX_IVCommon.h  OMX_Other.h  OMX_Types.h  OMX_Video.h
```

cgutman, 2022-01-07:
Hmm, it looks like it's finding the `bcm_host.h` header under `/usr/include` rather than `/opt/vc/include` where I'd expect it to be. Please attach CMakeOutput.log from `moonlight-embedded/build/CMakeFiles`.

Please also post the output of `dpkg -S /usr/include/bcm_host.h` and `dpkg -S /opt/vc/include/IL/OMX_Core.h`.

Is this Raspbian Buster?

esistgut, 2022-01-08:
This is on Raspberry OS Bullseye, not Buster.

```
pi@raspberrypi:~/moonlight-embedded/build $ dpkg -S /usr/include/bcm_host.h
libraspberrypi-dev: /usr/include/bcm_host.h

pi@raspberrypi:~/moonlight-embedded/build $ dpkg -S /opt/vc/include/IL/OMX_Core.h
dpkg-query: no path found matching pattern /opt/vc/include/IL/OMX_Core.h

pi@raspberrypi:~/moonlight-embedded/build $ md5sum /opt/vc/include/IL/OMX_Core.h
1ac724a2e37d854b76289ed31a15d30d  /opt/vc/include/IL/OMX_Core.h
```

[CMakeOutput.log](https://github.com/moonlight-stream/moonlight-embedded/files/7832436/CMakeOutput.log)

cgutman, 2022-01-08:
I pulled out my Raspberry Pi 1, flashed the latest Raspberry Pi OS Lite Bullseye image, and it builds (and runs) just fine for me. I don't have any `/opt/vc` directory present at all. That matches the behavior of of the Bullseye images on the other Raspberry Pis.

Since there's not a package on your system that provides those `/opt/vc` files, I suspect you either upgraded from an older version of Raspberry Pi OS which left the legacy `/opt/vc` files there or you installed something manually which put those files there.

If you want to try to make it work, you could try symlinking `/opt/vc/include/IL` to `/usr/include` by running `ln -s /opt/vc/include/IL /usr/include`. Another (more risky) option is to just nuke `/opt/vc` entirely.

esistgut, 2022-01-08:
You are right, something was wrong with my installation, even if I installed it with rpi-imager. This time I blanked the sd witth dd before and was able to compile moonlight-embedded.

closed 2022-01-08 by esistgut