moonlight-stream/moonlight-embedded #840: Video stream decoded correctly with lower bitrates, corrupted when using higher bitrate

camalolo, 2022-01-11:
Hello, this might be a hardware issue, but I am not sure.

Please provide the following info.

**_NVidia Geforce Experience version:_**

3.24.1.126

**Moonlight Embedded version:**

Master branch 2.5.2

**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._

https://github.com/moonlight-stream/moonlight-embedded

**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._

Odroid XU4

LAN is gigabit ethernet between the PC and the client

(odroid = 192.168.1.124)

ping 192.168.1.124
PING 192.168.1.124 (192.168.1.124) 56(84) bytes of data.
64 bytes from 192.168.1.124: icmp_seq=1 ttl=63 time=0.871 ms
64 bytes from 192.168.1.124: icmp_seq=2 ttl=63 time=0.820 ms
64 bytes from 192.168.1.124: icmp_seq=3 ttl=63 time=0.784 ms
64 bytes from 192.168.1.124: icmp_seq=4 ttl=63 time=0.786 ms
64 bytes from 192.168.1.124: icmp_seq=5 ttl=63 time=0.834 ms
64 bytes from 192.168.1.124: icmp_seq=6 ttl=63 time=0.719 ms
64 bytes from 192.168.1.124: icmp_seq=7 ttl=63 time=0.787 ms
64 bytes from 192.168.1.124: icmp_seq=8 ttl=63 time=0.813 ms
64 bytes from 192.168.1.124: icmp_seq=9 ttl=63 time=0.810 ms
64 bytes from 192.168.1.124: icmp_seq=10 ttl=63 time=0.767 ms
64 bytes from 192.168.1.124: icmp_seq=11 ttl=63 time=0.741 ms
64 bytes from 192.168.1.124: icmp_seq=12 ttl=63 time=0.805 ms
64 bytes from 192.168.1.124: icmp_seq=13 ttl=63 time=0.816 ms
64 bytes from 192.168.1.124: icmp_seq=14 ttl=63 time=0.833 ms
64 bytes from 192.168.1.124: icmp_seq=15 ttl=63 time=0.855 ms
64 bytes from 192.168.1.124: icmp_seq=16 ttl=63 time=0.761 ms
^C
--- 192.168.1.124 ping statistics ---
16 packets transmitted, 16 received, 0% packet loss, time 15578ms
rtt min/avg/max/mdev = 0.719/0.800/0.871/0.041 ms

iperf -c 192.168.1.124 -i1 -t20
------------------------------------------------------------
Client connecting to 192.168.1.124, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 172.29.241.89 port 40594 connected with 192.168.1.124 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0- 1.0 sec   113 MBytes   948 Mbits/sec
[  3]  1.0- 2.0 sec   110 MBytes   923 Mbits/sec
[  3]  2.0- 3.0 sec   106 MBytes   890 Mbits/sec
[  3]  3.0- 4.0 sec   109 MBytes   911 Mbits/sec
[  3]  4.0- 5.0 sec   110 MBytes   921 Mbits/sec
[  3]  5.0- 6.0 sec   111 MBytes   931 Mbits/sec
[  3]  6.0- 7.0 sec   112 MBytes   938 Mbits/sec
[  3]  7.0- 8.0 sec   109 MBytes   916 Mbits/sec
[  3]  8.0- 9.0 sec   112 MBytes   937 Mbits/sec
[  3]  9.0-10.0 sec   108 MBytes   906 Mbits/sec
[  3] 10.0-11.0 sec   111 MBytes   933 Mbits/sec
[  3] 11.0-12.0 sec   111 MBytes   932 Mbits/sec
[  3] 12.0-13.0 sec   112 MBytes   938 Mbits/sec
[  3] 13.0-14.0 sec   110 MBytes   924 Mbits/sec
[  3] 14.0-15.0 sec   112 MBytes   941 Mbits/sec
[  3] 15.0-16.0 sec   111 MBytes   928 Mbits/sec
[  3] 16.0-17.0 sec   107 MBytes   894 Mbits/sec
[  3] 17.0-18.0 sec   111 MBytes   928 Mbits/sec
[  3] 18.0-19.0 sec   111 MBytes   932 Mbits/sec
[  3] 19.0-20.0 sec   111 MBytes   930 Mbits/sec
[  3]  0.0-20.0 sec  2.15 GBytes   925 Mbits/sec

**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._

Ubuntu 20.04
Linux odroid 5.4.167-236 #1 SMP PREEMPT Tue Dec 21 09:46:48 EST 2021 armv7l armv7l armv7l GNU/Linux

**Verbose output `-verbose` of Moonlight Embedded:**

With -bitrate 4000 :

DISPLAY=:0 /opt/bin/moonlight -platform sdl -mapping /opt/share/moonlight/sdl.gamecontrollerdb.txt -1080 -bitrate 4000 -app "Desktop" stream -verbose
Moonlight Embedded 2.5.2 (CEC;SDL;X11;VDPAU;VAAPI;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.1.106...
GPU: NVIDIA GeForce RTX 3070, GFE: 3.24.0.126 (gs_04_37_30719632, 7.1.437.0)
Platform SDL2 (software decoding)
Stream 1920 x 1080, 60 fps, 4000 kbps
RTSP port: 48010
Initializing platform...done
Resolving host name...done
Initializing audio stream...done
Starting RTSP handshake...Audio port: 48000
Video port: 47998
Control port: 47999
done
Initializing control stream...done
Initializing video stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...Received first audio packet after 0 ms
Initial audio resync period: 660 milliseconds
done
Received first video packet after 0 ms
Stopping input stream...ENet wait interrupted
Control stream connection failed: -1
done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...ENet peer acknowledged disconnection
done
Cleaning up input stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up audio stream...done
Cleaning up platform...done

With -bitrate 12000

DISPLAY=:0 /opt/bin/moonlight -platform sdl -mapping /opt/share/moonlight/sdl.gamecontrollerdb.txt -1080 -bitrate 12000 -app "Desktop" stream -verbose
Moonlight Embedded 2.5.2 (CEC;SDL;X11;VDPAU;VAAPI;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.1.106...
GPU: NVIDIA GeForce RTX 3070, GFE: 3.24.0.126 (gs_04_37_30719632, 7.1.437.0)
Platform SDL2 (software decoding)
Stream 1920 x 1080, 60 fps, 12000 kbps
RTSP port: 48010
Initializing platform...done
Resolving host name...done
Initializing audio stream...done
Starting RTSP handshake...Audio port: 48000
Video port: 47998
Control port: 47999
done
Initializing control stream...done
Initializing video stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Received first audio packet after 0 ms
Initial audio resync period: 585 milliseconds
Received first video packet after 0 ms
Unrecoverable frame 314: 3+0=3 received < 14 needed
Network dropped 1 frame (frame 314)
Invalidate reference frame request sent (314 to 314)
Unrecoverable frame 326: 13+0=13 received < 14 needed
Network dropped 1 frame (frame 326)
Invalidate reference frame request sent (326 to 326)
Unrecoverable frame 338: 13+0=13 received < 14 needed
Network dropped 1 frame (frame 338)
Invalidate reference frame request sent (338 to 338)
Unrecoverable frame 343: 11+0=11 received < 14 needed
Network dropped 1 frame (frame 343)
Invalidate reference frame request sent (343 to 343)
Unrecoverable frame 346: 7+0=7 received < 14 needed
Network dropped 1 frame (frame 346)
Invalidate reference frame request sent (346 to 346)
Unrecoverable frame 351: 10+0=10 received < 13 needed
Network dropped 1 frame (frame 351)
Invalidate reference frame request sent (351 to 351)
Unrecoverable frame 356: 12+0=12 received < 14 needed
Network dropped 1 frame (frame 356)
Invalidate reference frame request sent (356 to 356)
Network dropped 1 frame (frame 376)
Invalidate reference frame request sent (376 to 376)
Unrecoverable frame 386: 7+0=7 received < 13 needed
Network dropped 2 frames (frames 386 to 387)
Invalidate reference frame request sent (386 to 387)
Unrecoverable frame 392: 11+0=11 received < 13 needed
Network dropped 1 frame (frame 392)
Invalidate reference frame request sent (392 to 392)
Unrecoverable frame 398: 4+0=4 received < 14 needed
Network dropped 1 frame (frame 398)

[... many more ...]

Invalidate reference frame request sent (4758 to 4759)
Unrecoverable frame 4764: 11+0=11 received < 14 needed
Network dropped 1 frame (frame 4764)
Invalidate reference frame request sent (4764 to 4764)
Unrecoverable frame 4769: 10+0=10 received < 14 needed
Network dropped 1 frame (frame 4769)
Invalidate reference frame request sent (4769 to 4769)
Connection is okay
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed: -1
done
Stopping video stream...done
Stopping control stream...ENet peer acknowledged disconnection
done
Cleaning up input stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up audio stream...done
Cleaning up platform...done

**What is the expected result?**

A beautiful video

This is the video with -bitrate 4000

https://photos.app.goo.gl/Z6Sp5UwCkrPL7r4G9

**What happens instead of that?**

A corrupted video

This is the video with -birate 12000

https://photos.app.goo.gl/Zy4KkGQyCdSyQiFV6

camalolo, 2022-01-11:
Could it be related to this ?

https://github.com/moonlight-stream/moonlight-qt/issues/490

cgutman, 2022-01-12:
The problem is your Odroid XU4's Ethernet adapter is choking on the UDP packet bursts sent from your PC. You can tell because normal drop patterns will leave some FEC packets intact (that's the +0 in all the drop messages). FEC packets always come after the frame data, so it's clearly dropping the tail of the frame every time.

You can often mitigate this by changing your host's Ethernet adapter to 100Mb, but apart from that there's nothing else you can do. The packets just don't make it to the networking stack at all. You can verify this in Wireshark if you want by looking at UDP 47998, decoding it as RTP, and noting the gaps in RTP sequence numbers.

camalolo, 2022-01-12:
Oh wow - thank you for such a precise answer ! Closing the issue.

closed 2022-01-12 by camalolo