moonlight-stream/moonlight-embedded #842: moonlight usage help commands

Albusaidi30, 2022-01-27:
Hello everyone

i have install moonlight embedded sucssesfully on raspbeary pi 3B+ buster version . i have appliaction installed in my gaming pc which collect all games from different launchers(EX: steam , Epic games , Xbox game pass and orign ) in one place , called playnite . what i need is to run moonlight with play playnite instead of steam big picture ..

i have see the moonlight help and [usage instruction](https://github.com/moonlight-stream/moonlight-embedded/wiki/Usage) but i did not understand. so i need help to write best setting commaned line for moonlight .

these steps what i have done:
1. go to NIVIDA Experience and add the playnite program into shield 
2. i run this command `moonlight list` to show the apps
3. the list is shown as 
`connect to 192.168.0.xxx` 
`1. playnite` 
`2. steam `

so i tried this command `moonlight stream -1080 -fps 60 -app playnite 192.168.0.xxx `

but did not work :(

so please help me who to write this kind of command , i need to rum moonlight in 1080p and 60 fps with playnite appliaction 
also please give as i giud how to use this help instruction to write a full commaned line 

 thanks

Albusaidi30, 2022-01-31:
problem solved. 
it was spelling mistake , the p(for playnite) letter was capital in the `moonlight list` but i am writing a small letter :(
so that why it did not work.

jgabriel98, 2022-03-13:
Good to know!
Could you close the issue now that it's solved?

to keep thing organized :D

closed 2022-07-23 by Albusaidi30