moonlight-stream/moonlight-embedded #845: Series X controller issues

Fantazzj, 2022-02-02:
Please provide the following info.

**_NVidia Geforce Experience version: latest
**Moonlight Embedded version: latest
**Moonlight Embedded source: Repository from guide
**Moonlight Embedded running on: Raspberry Pi 3B
**Moonlight Embedded running on distribution: Raspbian buster

Hi,
If I plug in my Xbox one controller, I think it is the first model, moonlight recognize it as Microsoft controller and it works without problems.
If I plug in my Xbox series X controller, moonlight recognize it as a generic Xbox controlle but it doesn't work out of the box.
Both controller were connected with working USB cables.
So I decided to crate a configuration using your command
this is the output:
`030000005e040000120b000009050000,Generic X-Box pad,a:b0,b:b1,x:b2,y:b3,start:b7,guide:b8,back:b6,leftstick:b9,rightstick:b10,leftshoulder:b4,rightshoulder:b5,dpup:h0.9,dpleft:h0.8,dpdown:h0.12,dpright:h0.2,leftx:a0,lefty:a1,rightx:a3,righty:a4,lefttrigger:a2,righttrigger:a5,platform:Linux`
I put it into a file and started moonlight, using that file as configuration.
Now everything on the gamepad is working, except for just the UP and the DOWN buttons on the D-PAD.
Already tried remapping but it didn't work.
Do you have any suggestion?

Fantazzj, 2022-03-04:
I solved the problem by downloading a configuration file, with:
`wget https://raw.githubusercontent.com/gabomdq/SDL_GameControllerDB/aecb6b0df26c20f83868111ae4c8b66649c59e69/gamecontrollerdb.txt`
and launching moonlight with:
`-mapping gamecontrollerdb.txt`

But I'll leave the issue open because I think it should work out of the box, or at least by mapping the controller with the integrated command.

jgabriel98, 2022-03-11:
for me `wget https://raw.githubusercontent.com/gabomdq/SDL_GameControllerDB/master/gamecontrollerdb.txt`
was the one, i guess they updated/fixed the file

cgutman, 2022-10-31:
Fixed with new mappings imported for v2.5.3

closed 2022-10-31 by cgutman