moonlight-stream/moonlight-embedded #847: Moonlight Embedded crashes after 10 minutes of playing on 64 bits version of Raspberry pi Bullseye

Cageman1984, 2022-02-09:
Hi!

I'm using moonlight version 2.5.0 for Raspbian/OSMC on a raspberry pi 3 with raspberry pi OS 64 (just became stable). Moonlight is installed from the bullseye repository and i've disable the default display driver as described in the [wiki](https://github.com/moonlight-stream/moonlight-embedded/wiki/Packages). I'm using an xbox one controller (model 1914) in combination with the [xpadneo](https://github.com/atar-axis/xpadneo) driver.

the client connects to an windows 10 PC with latest drivers (511.65) and GFE 3.25.0.84.

It works very well for 10 minutes and the crashes on the pi.

Are there known issues with the 64 bit version of raspberry pi bullseye?

For now I'm switching back to my Raspberry pi Bullseye 32 bit. That one seems to work just as well, but doesn't crash :).

cgutman, 2022-02-12:
Hmm, I definitely never saw that behavior, though I only tested the 64-bit version of the Pi 4. It's possible that there's some hardware or driver difference causing the crashes.

Cageman1984, 2022-02-22:
Tried my 64bit version again. There were some updates on both the client and host side, so i wanted to see if it fixed stuff. It didn't. This time I'll post some more info. It crashed and said:

assertion failure:/opt/rpi-userland/interface/mmal/vc/mmal_vc_client.c:204:mmal_vc_lookup_client_context():client_context_pool.contexts[CLIENT_CONTEXT_MAGIC_MASK(index)].inuse
Aborted

Does that tell you something? Do you need more info?

I do not actually need this version to work, but maybe you could use the info for development.