moonlight-stream/moonlight-embedded #852: Moonlight-embedded doesn't work witth sunshine and AMD Gpus anymore

citral23, 2022-06-26:
Herllo, as of 5fe7b36b4004c93d916e38183d58a2e9f6d5b9d6 and 26/06/22, moonlight embedded immediately crashes with a sole "Info: CLIENT DISCONNECTED" and/or "video buffer too small" weither the client is arm64/arm32 (Emuelec derivative) or an x86_64 laptop (Debian)

And also weither the sunshine server is running windows or debian (same machine)

However, moonlight-qt and moonlight-android work perfectly fine in every scenario.

It also works, if I plug an old Nvidia GTS5600 in the sunshine host instead of the AMD one.

It seems the problem is not amd model specific, as I have it with a 6700XT, a 5650G, and have a report of the same issue with a 5700XT.

Example on the x86_64 debian laptop : 

Moonlight Embedded 2.5.2 (SDL;X11;VDPAU;VAAPI;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.1.3...
GPU: , GFE: 3.23.0.74 (, 7.1.431.0)
Platform SDL2 (software decoding)
Stream 1280 x 720, 60 fps, 10000 kbps
RTSP port: 48010
Initializing platform...done
Resolving host name...done
Initializing audio stream...done
Starting RTSP handshake...Audio port: 48000
Video port: 47998
Control port: 47999
done
Initializing control stream...done
Initializing video stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...Received first video packet after 0 ms
done
Starting input stream...done
Received first audio packet after 4100 ms
Initial audio resync period: 500 milliseconds
Video decode buffer too small

cgutman, 2022-06-26:
Please try with 543dc087fce7473b87bbaef47a350c49f8aa3fd1 and see if that fixes it.

citral23, 2022-06-26:
Thank you, that fixes it for both the arm64 and the x86_64 clients completely :)

closed 2022-06-26 by citral23