moonlight-stream/moonlight-embedded #855: Can't stream on Pi3+ with Bullseye

shoulih, 2022-08-06:
Please provide the following info.

**_NVidia Geforce Experience version: 3.13.0.85
**Moonlight Embedded version:  2.5.2

Installed onto Bullseye with the instruction below:
https://github.com/moonlight-stream/moonlight-embedded/wiki/Packages

able to pair with the remote server, but can't stream

**Verbose output `-verbose` of Moonlight Embedded:**

Moonlight Embedded 2.5.2 (CEC;MMAL;ALSA;PULSE;EMBEDDED)
Connect to xxx.xxx.xxx.xxx...
GPU: Tesla M60, GFE: 3.13.0.85 (gs_04_05_23612337, 7.1.405.0)
Platform Raspberry Pi (Broadcom) MMAL
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Not mapping Logitech K400 Plus as a gamepad
libcec.so.6: cannot open shared object file: No such file or directory
libcec.so.6: cannot open shared object file: No such file or directory
Failed to initialize libcec interface
Can't find app Steam

Do I miss anything?

cgutman, 2022-08-06:
It looks like you didn't pass an `-app` parameter, so it just tried to launch Steam which failed.

Make sure you've added apps to stream (or enabled full desktop streaming) and try streaming using the specified app name. Instructions for doing that are [here](https://github.com/moonlight-stream/moonlight-docs/wiki/Setup-Guide#adding-custom-programs-that-are-not-automatically-found).

You can use the `moonlight list` command to get a list of all app names that are available for streaming.

shoulih, 2022-08-06:
this is the command I started moonlight:   "moonlight stream ip_address"

cgutman, 2022-08-07:
right, but you need to pass an `-app` parameter to `moonlight` if you want it to try to stream something else other than Steam Big Picture (which appears to not be working/installed on your system).