moonlight-stream/moonlight-embedded #857: is moonlight broken on ambernic devices after gfe update?

c0pE89, 2022-09-22:
Please provide the following info.

**_NVidia Geforce Experience version:_**
**Moonlight Embedded version:**
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?**

**What happens instead of that?**

cgutman, 2022-09-23:
Thanks for the report. 20a02acb07c56dfccf842020bb511e3fce57af0d should fix the issue.

Can you confirm that fixes it?

cgutman, 2022-10-31:
Closing as assumed fixed in v2.5.3

closed 2022-10-31 by cgutman