moonlight-stream/moonlight-embedded #858: Any plan to add hdr support?

benzhe, 2022-10-30:
Please provide the following info.

**_NVidia Geforce Experience version:_**
**Moonlight Embedded version:**
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?**

**What happens instead of that?**

cgutman, 2022-10-31:
The core code for HDR should be pretty trivial. The challenge is supporting the various video backends and testing on HDR-capable hardware.

What platform are you looking for HDR support for?

cgutman, 2022-11-09:
I introduced the basic HDR plumbing with c2f21b955d4d2dcccbf165c791db151024c24186 which enables HDR with the `-hdr` option.

On the Amlogic side, it looks like amcodec does most of the heavy lifting for HDR. I was able to successfully enter HDR mode on a Vero 4K with the latest code in master.

Which Amlogic device are you using?

benzhe, 2022-11-09:
> I introduced the basic HDR plumbing with [c2f21b9](https://github.com/moonlight-stream/moonlight-embedded/commit/c2f21b955d4d2dcccbf165c791db151024c24186) which enables HDR with the `-hdr` option.
> 
> On the Amlogic side, it looks like amcodec does most of the heavy lifting for HDR. I was able to successfully enter HDR mode on a Vero 4K with the latest code in master.
> 
> Which Amlogic device are you using?

That's cool! I will try it later. And my device is TX3, with S905x3 SOC.

benzhe, 2022-11-12:
Hi @TheChoconut , can you merge this commit into your fork? This request may seem weird since I found my compilation buggy on CoreELEC.