moonlight-stream/moonlight-embedded #859: Khadas Edge 2 installation issues

lofiPictures, 2022-11-07:
Hello, I am running ubuntu 22.04 on a khadas edge 2 SBC.

As per the Khadas community, I am running the following commands to install moonlight

```
sudo apt update
sudo apt-get install git libopus0 libexpat1 libasound2 libudev1 libavahi-client3 libcurl4 libevdev2 libenet7 libssl-dev libopus-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libevdev-dev libexpat1-dev libpulse-dev uuid-dev libenet-dev cmake gcc g++ fakeroot debhelper libavcodec-dev ffmpeg
git clone https://github.com/irtimmer/moonlight-embedded.git
cd moonlight-embedded
git submodule update --init
mkdir build
cd build/
cmake ../
make
sudo make install
sudo ldconfig
```
and at make, I'm getting the following error:

```
khadas@Khadas:~/moonlight-embedded/build$ make
[  2%] Building C object libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/AudioStream.c.o
[  4%] Building C object libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/ByteBuffer.c.o
[  6%] Building C object libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/Connection.c.o
[  8%] Building C object libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/ConnectionTester.c.o
[ 10%] Building C object libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/ControlStream.c.o
/home/khadas/moonlight-embedded/third_party/moonlight-common-c/src/ControlStream.c: In function ‘startControlStream’:
/home/khadas/moonlight-embedded/third_party/moonlight-common-c/src/ControlStream.c:1190:9: warning: implicit declaration of function ‘enet_address_set_address’; did you mean ‘enet_address_set_host’? [-Wimplicit-function-declaration]
 1190 |         enet_address_set_address(&address, (struct sockaddr *)&RemoteAddr, RemoteAddrLen);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
      |         enet_address_set_host
/home/khadas/moonlight-embedded/third_party/moonlight-common-c/src/ControlStream.c:1191:9: warning: implicit declaration of function ‘enet_address_set_port’; did you mean ‘enet_address_set_host’? [-Wimplicit-function-declaration]
 1191 |         enet_address_set_port(&address, ControlPortNumber);
      |         ^~~~~~~~~~~~~~~~~~~~~
      |         enet_address_set_host
/home/khadas/moonlight-embedded/third_party/moonlight-common-c/src/ControlStream.c:1194:42: error: ‘ENetAddress’ has no member named ‘address’
 1194 |         client = enet_host_create(address.address.ss_family, NULL, 1, 1, 0, 0);
      |                                          ^
/home/khadas/moonlight-embedded/third_party/moonlight-common-c/src/ControlStream.c:1194:62: warning: passing argument 2 of ‘enet_host_create’ makes integer from pointer without a cast [-Wint-conversion]
 1194 |        client = enet_host_create(address.address.ss_family, NULL, 1, 1, 0, 0);
      |                                                             ^~~~
      |                                                             |
      |                                                             void *

In file included from /home/khadas/moonlight-embedded/third_party/moonlight-common-c/src/Limelight-internal.h:14,
                 from /home/khadas/moonlight-embedded/third_party/moonlight-common-c/src/ControlStream.c:1:
/usr/include/enet/enet.h:545:60: note: expected ‘size_t’ {aka ‘long unsigned int’} but argument is of type ‘void *’
  545 | ENET_API ENetHost * enet_host_create (const ENetAddress *, size_t, size_t, enet_uint32, enet_uint32);
      |                                                            ^~~~~~
/home/khadas/moonlight-embedded/third_party/moonlight-common-c/src/ControlStream.c:1194:18: error: too many arguments to function ‘enet_host_create’
 1194 |         client = enet_host_create(address.address.ss_family, NULL, 1, 1, 0, 0);
      |                  ^~~~~~~~~~~~~~~~
In file included from /home/khadas/moonlight-embedded/third_party/moonlight-common-c/src/Limelight-internal.h:14,
                 from /home/khadas/moonlight-embedded/third_party/moonlight-common-c/src/ControlStream.c:1:
/usr/include/enet/enet.h:545:21: note: declared here
  545 | ENET_API ENetHost * enet_host_create (const ENetAddress *, size_t, size_t, enet_uint32, enet_uint32);
      |                     ^~~~~~~~~~~~~~~~
make[2]: *** [libgamestream/CMakeFiles/moonlight-common.dir/build.make:132: libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/ControlStream.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:176: libgamestream/CMakeFiles/moonlight-common.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

```

I'v been trying to get this to work for a few days now, and tried many different things but always end up with this same error?

Maybe someone has encountered a similar issue?

Thanks for any help!

cgutman, 2022-11-08:
The problem is that you're building with `libenet-dev` instead of the fork of `enet` included in the `moonlight-common-c` submodule.

Run `git submodule update --init --recursive` and see if that fixes it. If not, try removing the `libenet-dev` package.

lofiPictures, 2022-11-08:
Thanks for the help cgutman, this got me further along in the make process, but it still chocked around 96%... Here is the log, and I checked in synaptic, libenet-dev is not installed. Any other thoughts? Really appreciate the help!

```
khadas@Khadas:~/moonlight-embedded/build$ cmake ../
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- Found ALSA: /usr/lib/aarch64-linux-gnu/libasound.so (found version "1.2.6.1") 
-- Found Opus: /usr/lib/aarch64-linux-gnu/libopus.so  
-- Could NOT find Broadcom-OMX (missing: OPENMAXIL_LIBRARY) 
-- Could NOT find Freescale (missing: FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY) 
-- Could NOT find Amlogic (missing: AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY) 
-- Could NOT find Rockchip (missing: ROCKCHIP_INCLUDE_DIR ROCKCHIP_LIBRARY) 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libevdev'
--   Found libevdev, version 1.12.1
-- Checking for module 'libudev'
--   Found libudev, version 249
-- Checking for module 'sdl2>=2.0.4'
--   No package 'sdl2' found
-- Checking for module 'libavcodec'
--   Found libavcodec, version 58.134.100
-- Checking for module 'libavutil'
--   Found libavutil, version 56.70.100
-- Checking for module 'vdpau'
--   Found vdpau, version 1.4
-- Checking for module 'libva'
--   Found libva, version 1.14.0
-- Checking for module 'egl'
--   No package 'egl' found
-- Checking for module 'glesv2'
--   No package 'glesv2' found
-- Checking for module 'x11'
--   Found x11, version 1.7.5
-- Checking for module 'libva-x11'
--   Found libva-x11, version 1.14.0
-- Checking for module 'libpulse-simple'
--   Found libpulse-simple, version 15.99.1
-- Checking for module 'libcec>=4'
--   No package 'libcec' found
-- Checking for module 'mmal'
--   Found mmal, version 1
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (LibUUID).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  cmake/FindLibUUID.cmake:15 (INCLUDE)
  libgamestream/CMakeLists.txt:3 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for one of the modules 'uuid'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found CURL: /usr/lib/aarch64-linux-gnu/libcurl.so (found version "7.81.0")  
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found suitable version "3.0.2", minimum required is "1.0.2")  
-- Found EXPAT: /usr/lib/aarch64-linux-gnu/libexpat.so (found version "2.4.7") 
-- Checking for module 'avahi-client'
--   Found avahi-client, version 0.8
-- Configuring done
-- Generating done
[... 2343 lines cut ...]
      |                      int
      |                     %ld
/home/khadas/moonlight-embedded/third_party/h264bitstream/h264_stream.c:2705:20: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘long int’ [-Wformat=]
 2705 |             printf("%d.%d: ", b->p - b->start, b->bits_left); sh->pwt.luma_offset_l0[ i ] = bs_read_se(b); printf("sh->pwt.luma_offset_l0[ i ]: %d \n", sh->pwt.luma_offset_l0[ i ]);
      |                    ^~~~~~~~~  ~~~~~~~~~~~~~~~
      |                                    |
      |                                    long int
/home/khadas/moonlight-embedded/third_party/h264bitstream/h264_stream.c:35:77: note: in definition of macro ‘printf’
   35 | ..) fprintf((h264_dbgfile == NULL ? stdout : h264_dbgfile), __VA_ARGS__)
      |                                                             ^~~~~~~~~~~

/home/khadas/moonlight-embedded/third_party/h264bitstream/h264_stream.c:2705:22: note: format string is defined here
 2705 |             printf("%d.%d: ", b->p - b->start, b->bits_left); sh->pwt.luma_offset_l0[ i ] = bs_read_se(b); printf("sh->pwt.luma_offset_l0[ i ]: %d \n", sh->pwt.luma_offset_l0[ i ]);
      |                     ~^
      |                      |
      |                      int
      |                     %ld
/home/khadas/moonlight-embedded/third_party/h264bitstream/h264_stream.c:2709:20: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘long int’ [-Wformat=]
 2709 |             printf("%d.%d: ", b->p - b->start, b->bits_left); sh->pwt.chroma_weight_l0_flag[i] = bs_read_u1(b); printf("sh->pwt.chroma_weight_l0_flag[i]: %d \n", sh->pwt.chroma_weight_l0_flag[i]);
      |                    ^~~~~~~~~  ~~~~~~~~~~~~~~~
      |                                    |
      |                                    long int
/home/khadas/moonlight-embedded/third_party/h264bitstream/h264_stream.c:35:77: note: in definition of macro ‘printf’
   35 | ..) fprintf((h264_dbgfile == NULL ? stdout : h264_dbgfile), __VA_ARGS__)
      |                                                             ^~

cgutman, 2022-11-08:
Hmm, I'm not sure why CMake is finding MMAL. That's a library for decoding on the Raspberry Pi.

What is the expected hardware video decoding API for this device? Is there a custom build of FFmpeg or some special vendor-specific API?

lofiPictures, 2022-11-08:
I might have some pi packages installed by mistake, I've been fooling around with things on my linux build... For the decode api, from what I can gather, the edge 2 supports GStreamer, as well as mppvideodec, maybe others. It's early days on the edge 2, so there isn't that much support yet. But this thread seems helpful:

https://forum.khadas.com/t/mali-g610-hw-decode-encode/17300

Thanks again, I'll try removing pi packages.

lofiPictures, 2022-11-08:
So I removed libraspberrypi-dev, re-ran the commands and now I'm getting this at make:

```
khadas@Khadas:~/moonlight-embedded/build$ cmake ../
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- Found ALSA: /usr/lib/aarch64-linux-gnu/libasound.so (found version "1.2.6.1") 
-- Found Opus: /usr/lib/aarch64-linux-gnu/libopus.so  
-- Could NOT find Broadcom-OMX (missing: BROADCOM_INCLUDE_DIR VCOS_LIBRARY VCHIQ_LIBRARY OPENMAXIL_LIBRARY BCM_HOST_LIBRARY) 
-- Could NOT find Freescale (missing: FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY) 
-- Could NOT find Amlogic (missing: AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY) 
-- Could NOT find Rockchip (missing: ROCKCHIP_INCLUDE_DIR ROCKCHIP_LIBRARY) 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libevdev'
--   Found libevdev, version 1.12.1
-- Checking for module 'libudev'
--   Found libudev, version 249
-- Checking for module 'sdl2>=2.0.4'
--   No package 'sdl2' found
-- Checking for module 'libavcodec'
--   Found libavcodec, version 58.134.100
-- Checking for module 'libavutil'
--   Found libavutil, version 56.70.100
-- Checking for module 'vdpau'
--   Found vdpau, version 1.4
-- Checking for module 'libva'
--   Found libva, version 1.14.0
-- Checking for module 'egl'
--   No package 'egl' found
-- Checking for module 'glesv2'
--   No package 'glesv2' found
-- Checking for module 'x11'
--   Found x11, version 1.7.5
-- Checking for module 'libva-x11'
--   Found libva-x11, version 1.14.0
-- Checking for module 'libpulse-simple'
--   Found libpulse-simple, version 15.99.1
-- Checking for module 'libcec>=4'
--   No package 'libcec' found
-- Checking for module 'mmal'
--   No package 'mmal' found
-- Could NOT find MMAL (missing: BROADCOM_INCLUDE_DIR VCOS_LIBRARY MMAL_CORE_LIBRARY MMAL_UTIL_LIBRARY MMAL_VC_CLIENT_LIBRARY BCM_HOST_LIBRARY) 
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (LibUUID).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  cmake/FindLibUUID.cmake:15 (INCLUDE)
  libgamestream/CMakeLists.txt:3 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for one of the modules 'uuid'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found CURL: /usr/lib/aarch64-linux-gnu/libcurl.so (found version "7.81.0")  
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found suitable version "3.0.2", minimum required is "1.0.2")  
-- Found EXPAT: /usr/lib/aarch64-linux-gnu/libexpat.so (found version "2.4.7") 
-- Checking for module 'avahi-client'
--   Found avahi-client, version 0.8
CMake Error at CMakeLists.txt:200 (message):
  No video output available

-- Configuring incomplete, errors occurred!
See also "/home/khadas/moonlight-embedded/build/CMakeFiles/CMakeOutput.log".
```

lofiPictures, 2022-11-09:
Once again thank you for your valuable insight. But I fear this is starting to get out of my comfort zone... haha! I will transfer the information to the Khadas forums, hopefully the kind folks there can make this all easier for people like me in future distros. On a side note, I was able to install Moonlight with flatpack, but as soon as I booted it up, it complained that it couldn't find a video decoder for my hardware.. I guess it's all kind of the same issue. It then switched to software decoding. It all worked, just not in a usable way. Also, the edge 2 has an android distro that can run moonlight. It performs pretty well, but I'd rather get it working on Linux. Anyway, maybe this information can help some people out in the future ;)

Cheers

migenbrawl, 2023-10-25:
I encountered problems installing moonlight with the same commands, but was able to make them work on an Edge2 in Ubuntu after installing the packages libdrm-dev and rockchip-mpp-dev packages.