moonlight-stream/moonlight-embedded #861: failed to stream on rk3588 sbc

amazingfate, 2022-12-07:
I have a radxa rock5b rk3588 SBC, and I've configured a working hardware decoding environment according to this post: https://forum.armbian.com/topic/24802-kodi-for-rk35xx-510-legacy-kernel/.
To make ffmpeg work I add this patch:
```
diff --git a/src/video/ffmpeg.c b/src/video/ffmpeg.c
index 6f8f899..48c9974 100644
--- a/src/video/ffmpeg.c
+++ b/src/video/ffmpeg.c
@@ -61,9 +61,9 @@ int ffmpeg_init(int videoFormat, int width, int height, int perf_lvl, int buffer

   ffmpeg_decoder = perf_lvl & VAAPI_ACCELERATION ? VAAPI : SOFTWARE;
   if (videoFormat & VIDEO_FORMAT_MASK_H264) {
-    decoder = avcodec_find_decoder_by_name("h264");
+    decoder = avcodec_find_decoder_by_name("h264_rkmpp");
   } else if (videoFormat & VIDEO_FORMAT_MASK_H265) {
-    decoder = avcodec_find_decoder_by_name("hevc");
+    decoder = avcodec_find_decoder_by_name("hevc_rkmpp");
   } else {
     printf("Video format not supported\n");
     return -1;
diff --git a/src/video/rk.c b/src/video/rk.c
index be355be..434489c 100644
--- a/src/video/rk.c
+++ b/src/video/rk.c
@@ -44,7 +44,7 @@
 #define RK_H265 16777220

 #ifndef DRM_FORMAT_NV12_10
-#define DRM_FORMAT_NV12_10 fourcc_code('N', 'A', '1', '2')
+#define DRM_FORMAT_NV12_10 fourcc_code('N', 'V', '1', '5')
 #endif

 void *pkt_buf = NULL;
```
When I run with command `moonlight stream 10.5.120.26 -app Desktop`, I got this output:
```
Connecting to 10.5.120.26...
RTSP port: 48010
Initializing platform...done
Resolving host name...done
Initializing audio stream...done
Starting RTSP handshake...Reference frame invalidation is not supported by this host
Audio port: 48000
Video port: 47998
Control port: 47999
done
Initializing control stream...done
Initializing video stream...done
Initializing input stream...done
Starting control stream...done
moonlight: /home/jfliu/dev/moonlight/moonlight-embedded/src/video/rk.c:359: rk_setup: Assertion `plane_id' failed.
Aborted
```
If I add option `-platform x11`, I can get stream successfully. I can see from the output of `cat /proc/mpp_service/sessions-summary` that the hardware decoding is working.
What I want to know is, what's the difference between platform rk and x11? It seems that rk uses drm to show the stream which should be better than x11. And how to make platform rk works?

amazingfate, 2022-12-08:
After this patch I can use platform rk now.
```
diff --git a/src/video/ffmpeg.c b/src/video/ffmpeg.c
index 6f8f899..48c9974 100644
--- a/src/video/ffmpeg.c
+++ b/src/video/ffmpeg.c
@@ -61,9 +61,9 @@ int ffmpeg_init(int videoFormat, int width, int height, int perf_lvl, int buffer

   ffmpeg_decoder = perf_lvl & VAAPI_ACCELERATION ? VAAPI : SOFTWARE;
   if (videoFormat & VIDEO_FORMAT_MASK_H264) {
-    decoder = avcodec_find_decoder_by_name("h264");
+    decoder = avcodec_find_decoder_by_name("h264_rkmpp");
   } else if (videoFormat & VIDEO_FORMAT_MASK_H265) {
-    decoder = avcodec_find_decoder_by_name("hevc");
+    decoder = avcodec_find_decoder_by_name("hevc_rkmpp");
   } else {
     printf("Video format not supported\n");
     return -1;
diff --git a/src/video/rk.c b/src/video/rk.c
index be355be..f0dd191 100644
--- a/src/video/rk.c
+++ b/src/video/rk.c
@@ -44,7 +44,7 @@
 #define RK_H265 16777220

 #ifndef DRM_FORMAT_NV12_10
-#define DRM_FORMAT_NV12_10 fourcc_code('N', 'A', '1', '2')
+#define DRM_FORMAT_NV12_10 fourcc_code('N', 'V', '1', '5')
 #endif

 void *pkt_buf = NULL;
@@ -344,7 +344,7 @@ int rk_setup(int videoFormat, int width, int height, int redrawRate, void* conte
         if (!prop) {
           continue;
         }
-        if (!strcmp(prop->name, "type") && props->prop_values[j] == DRM_PLANE_TYPE_OVERLAY) {
+        if (!strcmp(prop->name, "type") && props->prop_values[j] == DRM_PLANE_TYPE_PRIMARY) {
           plane_id = ovr->plane_id;
         }
         drmModeFreeProperty(prop);
```

lsartory, 2022-12-30:
Confirmed, it fixed the 'rk' platform on my NanoPi R6S as well.
I guess this cannot be applied as-is though, as it breaks other hardware?

AquaWolf, 2023-01-02:
I'm also trying to get rk platform working with a rock5b do I have to apply both patches? I'm getting /src/video/rk.c:379: rk_setup: Assertion `!ret' failed.

lsartory, 2023-01-02:
You should only apply the patch from the second post as it contains the changes from the first one.
Oddly enough, it seems that the error is different depending on the system. I didn't get the assertion failed error, but my screen remained blank.

zylo117, 2023-01-13:
@amazingfate  this part of patch makes it work. There is no need to change decoder from mpp to ffmpeg. rk3588 is faster when using direct mpp decoding instead of ffmpeg-wrapped mpp
```
@@ -344,7 +344,7 @@ int rk_setup(int videoFormat, int width, int height, int redrawRate, void* conte
         if (!prop) {
           continue;
         }
-        if (!strcmp(prop->name, "type") && props->prop_values[j] == DRM_PLANE_TYPE_OVERLAY) {
+        if (!strcmp(prop->name, "type") && props->prop_values[j] == DRM_PLANE_TYPE_PRIMARY) {
           plane_id = ovr->plane_id;
         }
         drmModeFreeProperty(prop);
```

lsartory, 2023-01-14:
@zylo117 Indeed I have tested with only this change and it also works, thanks.
I couldn't really tell any performance difference so far, though.

longcat99, 2023-01-19:
> @zylo117事实上，我只测试了这个变化，它也有效，谢谢。 不过，到目前为止，我真的无法说出任何性能差异。

Connecting to 192.168.100.100...
RTSP port: 48010
Initializing platform...done
Resolving host name...done
Initializing audio stream...done
Starting RTSP handshake...Audio port: 48000
Video port: 47998
Control port: 47999
done
Initializing control stream...done
Initializing video stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Received first audio packet after 0 ms
Initial audio resync period: 750 milliseconds
Received first video packet after 0 ms
moonlight: /home/longcat/moonlight-embedded/src/video/rk.c:112: display_thread: Assertion `!ret' failed.
已放弃 (核心已转储)

lsartory, 2023-01-19:
I think there are actually two different issues here, because I never had the 'assertion failed' error, for me it was just that nothing was being displayed without the patch mentioned by @zylo117.
It probably depends on the software environment as well (mpp version for example)...

closed 2023-03-06 by cgutman

amazingfate, 2023-03-06:
For rk356x and rk3588 devices running 5.10 legacy kernel, `CONFIG_DRM_IGNORE_IOTCL_PERMIT` has to be enabled to run moonlight-embedded, otherwise you have to use moonlight-qt instead.

commit 6ab37d00 (2023-03-06): Accept both primary and overlay planes on Rockchip platforms

Fixes #861