moonlight-stream/moonlight-embedded #864: Streaming not working with error 'Waiting for IDR frame' and 'Connection is poor' error message

marcelhoelscher, 2023-02-18:
Please provide the following info.

**NVidia Geforce Experience version: 3.27.0.112**
**Moonlight Embedded version: 2.5.3**
**Moonlight Embedded source:** repository:  https://dl.cloudsmith.io/public/moonlight-game-streaming/moonlight-embedded/deb/raspbian buster InRelease
**Moonlight Embedded running on:** _Raspberry Pi 4_
**Moonlight Embedded running on distribution:** _Raspbian with RetroPie installed on it_

**Verbose output `-verbose` of Moonlight Embedded: see attached console output**
[console_output.log](https://github.com/moonlight-stream/moonlight-embedded/files/10774965/console_output.log)

**What is the expected result? Game streaming works**

**What happens instead of that? Game streaming does not work, see console output above**

**Further description**
I am streaming from my Win 10 PC to my Raspberry Pi 4. This worked for the last 12 months without any flaw. The last time i played via moonlight was three weeks ago. Yesterday i tried again and it ended with the above console ouput.

The console output says, that my lan connection is poor. But that can't be: I tested it and it is working fine at ca 1 GBit. I did not change anything on the raspberry and also i did not touch my network configuration.

Only thing i changed was to prepare my PC for Win 11 (means TPM 2.0 support, safe boot etc). Maybe i also updated the NVidia-Software, but i am not quite sure. 

In order to solve the issue, i updated the os of my Raspberry including Moonlight, also the NVidia-Software on my Win 10 PC. 

Any help is appreciated.

W4stedWizard, 2023-02-21:
Same issue here, 
GFE 3.27.0.112
GRD 528.49
ML-embedded 2.5.2-master-b9703e7
Raspi3B running Raspbian Bullseye
Windows 10 with the latest updates as of posting

I think this coincided with a GFE update or a Nvidia Driver Update - it happened basically one day to the next.

marcelhoelscher, 2023-02-21:
Seems like something similar happened  a few years ago:

https://github.com/moonlight-stream/moonlight-embedded/issues/447#issuecomment-267759416

At that time it was caused by a change in the nvidia shield protocol.

W4stedWizard, 2023-02-21:
Just realised I had some issues updating moonlight, updated to 2.5.3, replaced this issue with a connection Segfault.
Unsure which is worse.
Using QT as a workaround for now, assuming a fix will come or it's a localised issue on my device

rona1707, 2023-02-24:
same issue here:

GFE 3.27.0.112
GRD 528.49

Windows 10 Pro
Raspberry Pi3b+

Moonlight 2.5.3

seems to be an issue with the GFE Update

Interein2, 2023-03-02:
Same issue: 

Raspberry Pi3b+
Win10Pro

It was working till last Geforce Experience update (until I oppened GFE to check and it updated to 3.27.0.112)

Moonlight-Qt is working (Rpi3B+ is hard on it)

marcelhoelscher, 2023-03-02:
We should not be too hopeful that this issue gets a fix. NVidia dropped support for their Gamestream feature, so "Nvidia native" streaming to the moonlight client has no future. See this article for details: 

https://github.com/moonlight-stream/moonlight-docs/wiki/NVIDIA-GameStream-End-Of-Service-Announcement-FAQ

I will give the sunshine streaming host a try. The good point in this: I am no longer  bound to Nvidia in order to stream my games to my TV.

marcelhoelscher, 2023-03-03:
Not sure if this is possible or desirable. 

Maybe you wanna try sunshine. I did today and was surprised how good it performed. Installation on the host PC is trivial and you can configure it as a service  that is started together with Windows and running in the background.

I will test it further dringend t

rona1707, 2023-03-03:
@marcelhoelscher  

Did you found any guide how to setup sunshine application? I've got 2 application on first start of Sunshine (Desktop, Steam Big Picture), but the properties (e.g Output, Command, Working Directory, ... are empty

marcelhoelscher, 2023-03-03:
Here is a link to the official documentation:
 https://docs.lizardbyte.dev/projects/sunshine/en/latest/

Most interesting may be the "Usage" section: https://docs.lizardbyte.dev/projects/sunshine/en/latest/about/usage.html

The existing two configs are supposed to work out of the Box. You can add additional games there by giving the path to the corresponding .exe.

marcelhoelscher, 2023-03-03:
sorry closed this issue by mistake

CaptainArni, 2023-03-04:
I just tried out sunshine and it is working really well. I didn't notice a difference between gamestream and sunshine when gaming.
The only thing that does not seem to work is watching Netflix / Amazon Prime. Audio plays but there is a black screen only. This happens with the moonlight client on pi and I also tested with Android, same problem.

rona1707, 2023-03-08:
Thx @marcelhoelscher 

I tried sunshine and managed to get it working with moonlight on my Pi3. The pairing process wasn't quite obvious, as I needed to unpair for GFE and re-pair for sunshine. But at the end it worked out well and the performance is fine after some small adaptions. The Desktop application worked out of the box. In GFE I needed to set up the profile with the "mstsc.exe" to stream the Desktop. I basically used the Desktop application and started my PC games there, which was sufficient enough.

marcelhoelscher, 2023-03-08:
@rona1707: Glad, that i could help.

I also tried it out a bit more. Sunshine is a cool piece of software. I am very happy, that Sunshine enables me to also stream from my Linux installation.

Can you tell me which performance optimization you made?

rona1707, 2023-03-10:
I messed around with the NVENC encoder settings and the quantization parameter in the "Advanced" configuration tab. It felt a bit handier reducing the quality in terms of gaining lower latency.