moonlight-stream/moonlight-embedded #865: Can't redirect stdout

christovic, 2023-02-27:
Please provide the following info.

**_NVidia Geforce Experience version:_**
N/A
**Moonlight Embedded version:**
Moonlight Embedded 2.5.3-master-4c58ac9
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
Compiled from source
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
x86 Intel 12th Gen
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
Debian

Strange one, I can't redirect stdout...?

Using bash, the command

```
moonlight stream -app Desktop -mapping gamecontrollerdb.txt.1 -1080 -fps 120 -bitrate 80000 -codec h265
```

will produce "Searching for server..." as output

but

```
moonlight stream -app Desktop -mapping gamecontrollerdb.txt.1 -1080 -fps 120 -bitrate 80000 -codec h265 > stdout.test
```

will create the file, but there'll be no output in it. There is no output in the terminal, so I know that it's definitely stdout.

I can't see anything special in the printf statements in main.c.

What is more curious, is that 

```
moonlight > stdout.test
```
will write the stdout of the help text to the file.

Not a huge issue, but I'm struggling to figure this one out.

christovic, 2023-02-27:
Okay, so it does send it to stdout, but only once it quits. If I interrupt execution, it doesn't. I can't figure out why it wouldn't print to the file at the same time as it does to terminal though.

cgutman, 2023-02-28:
It's because stdout is line-buffered when writing to a TTY but fully-buffered when writing to a file.

There are a few utilities that can adjust the buffering method of stdout: https://stackoverflow.com/questions/11337041/force-line-buffering-of-stdout-in-a-pipeline

christovic, 2023-02-28:
Amazing, thank you! I knew it would be something that went over my head.