moonlight-stream/moonlight-embedded #868: "plane_id" assertion failure on rk3399

eigenbrot, 2023-05-01:
Please provide the following info.

**_NVidia Geforce Experience version:_ NA**
**Moonlight Embedded version:** _built from source @ commit 5bb47c_
**Moonlight Embedded source:** _https://github.com/moonlight-stream/moonlight-embedded_
**Moonlight Embedded running on:** _RockPi 4b_
**Moonlight Embedded running on distribution:** _Armbian Bullseye_

**Verbose output `-verbose` of Moonlight Embedded:**
```
Moonlight Embedded 2.5.3 (ROCKCHIP;SDL;X11;VDPAU;VAAPI;ALSA;PULSE;EMBEDDED)
Connecting to xx.xxx.xxx.xx..
GPU: , GFE: 3.23.0.74 (, 7.1.431.0)
Platform Rockchip VPU
Loading mappingfile ../third_party/SDL_GameControllerDB/gamecontrollerdb.txt
Not mapping Logitech K400 Plus as a gamepad
Stream 1280 x 720, 60 fps, 10000 kbps
RTSP port: 48010
Initializing platform...done
Resolving host name...done
Packet size capped at 1KB for remote streaming
Initializing audio stream...done
Starting RTSP handshake...Reference frame invalidation is not supported by this host
Audio port: 48000
Video port: 47998
Control port: 47999
done
Initializing control stream...done
Initializing video stream...done
Initializing input stream...done
Starting control stream...done
moonlight: /home/luna/moonlight-embedded/src/video/rk.c:488: rk_setup: Assertion `plane_id' failed.
Aborted
```

**What is the expected result?**
Moonlight runs

**What happens instead of that?**
Moonlight aborts

This issue is perhaps similar to #861, but the code I'm trying to build already has [that fix](https://github.com/moonlight-stream/moonlight-embedded/commit/6ab37d00364e99e37442ba4aaaa1beabcc275b25) merged in, so there's something else going on here.

In order to find the rockchip libraries I had to install [mpp from rockchip-linux](https://github.com/rockchip-linux/mpp). If I should have looked somewhere else please let me know.

Any help is appreciated. Thanks!

voidvisor, 2023-05-07:
Experiencing the same issue on RK3588S
NVidia Geforce Experience version: NA (sunshine)
Moonlight Embedded version: built from source @ commit 5bb47c
Moonlight Embedded source: https://github.com/moonlight-stream/moonlight-embedded
Moonlight Embedded running on: Orange Pi 5B
Moonlight Embedded running on distribution: Orange Pi Debian 11 Bullseye

Just like in #861 using `-platform x11` works, but for some reason produces washed-out colors for me. It may not be related to x11.

eigenbrot, 2023-05-07:
Further research into this problem suggest that it's an issue with the underlying drivers available for the rk3399. This is not a moonlight issues so I'm going to close this.

@voidvisor I would suggest looking at [this set of packages](https://forum.radxa.com/t/introduction-to-rockchip-multimedia-ppa-for-ubuntu-jammy/14537) that seem to have been tested specifically for the rk3588.

closed 2023-05-07 by eigenbrot