moonlight-stream/moonlight-embedded #873: Amlogic platform issue

flyingataildragger, 2023-07-31:
Please provide the following info.

**_NVidia Geforce Experience version:_** Current version
**Moonlight Embedded version:** 2.5.3
**Moonlight Embedded source:** GitHub
**Moonlight Embedded running on:** Le Potato
**Moonlight Embedded running on distribution:** Raspbian Buster

**Verbose output `-verbose` of Moonlight Embedded:**
```
pi@raspberrypi:~ $ moonlight --verbose stream
Moonlight Embedded 2.5.3 (PI;MMAL;ALSA;PULSE;EMBEDDED)
Searching for server...
Connecting to 192.168.1.72...
GPU: NVIDIA GeForce RTX 3060, GFE: 3.27.0.112 (gs_04_50_32193813, 7.1.450.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share/moonlight/gamecontrollerdb.txt
Not mapping meson_ao_cec as a gamepad
Not mapping meson-ir as a gamepad
Not mapping Lite-On Technology Corp HP USB Multimedia Keyboard as a gamepad
Not mapping Lite-On Technology Corp HP USB Multimedia Keyboard Consumer Control as a gamepad
Not mapping Lite-On Technology Corp HP USB Multimedia Keyboard System Control as a gamepad
Not mapping SIGMACHIP Usb Mouse as a gamepad
Stream 1280 x 720, 60 fps, 10000 kbps
RTSP port: 48010
Initializing platform...done
Resolving host name...done
Initializing audio stream...done
Starting RTSP handshake...Audio port: 48000
Video port: 47998
Control port: 47999
done
Initializing control stream...done
Initializing video stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...* failed to open vchiq instance
```
**What is the expected result?**

My stream should begin

**What happens instead of that?**

Steam is launched on host PC but I am getting that `failed to open vchiq instance`. I see there is an AmLogic platform for Moonlight however when I run `moonlight stream -platform aml` it kicks back `Platform 'aml' not found`. Are these two problems related? How do I get the aml platform? Is there a better method to running Moonlight on an S905X chip? I've spent days researching this and everything I've come up with is from years ago where functionality has changed or unrelated to my specific problem. Thanks in advance.

zjoasan, 2023-09-01:
I only have experience from OSMC and the Vero4(+), where we compiled it but had to exchange some libs. See the [wiki](https://github.com/moonlight-stream/moonlight-embedded/wiki/Full-details-for-Compilation-and-Installation-in-Raspbian-Or-OSMC). Since you are using another OS i don't know the unique names for the same libs in your OS.

zjoasan, 2024-04-12:
Another thing i notice that somehow the installer, even after ldconfig and reboot didn't find the libs. So I made this filestructure:
```
~/moonlight-embedded/
keys/
libmoonlight-aml.so
moonlight
moonlight.conf
```