moonlight-stream/moonlight-embedded #875: rk_setup: Assertion 'plane_id' failed. when starting Stream

MarkTwain007, 2023-09-02:
Please provide the following info.

**_NVidia Geforce Experience version:_** Sunshine Version 0.20.0
**Moonlight Embedded version:** v2.6.0
**Moonlight Embedded source:**  GitHub
**Moonlight Embedded running on:** OrangePi 5 4GB
**Moonlight Embedded running on distribution:** Debian Bullseye
**Verbose output `-verbose` of Moonlight Embedded:**
orangepi@orangepi5:~$ moonlight --verbose stream -app Desktop 192.168.0.136
Moonlight Embedded 2.6.0 (ROCKCHIP;X11;ALSA;PULSE;EMBEDDED)
Connecting to 192.168.0.136...
GPU: , GFE: 3.23.0.74 (, 7.1.431.-1)
Server codec flags: 0xf03
Platform Rockchip VPU
Loading mappingfile /usr/local/share//moonlight/gamecontrollerdb.txt
Not mapping adc-keys as a gamepad
Not mapping rockchip-dp0 rockchip-dp0 as a gamepad
Not mapping headset-keys as a gamepad
Not mapping rockchip-es8388 Headset as a gamepad
Not mapping Logitech USB Receiver as a gamepad
Not mapping Logitech USB Receiver Mouse as a gamepad
Not mapping Logitech USB Receiver Consumer Control as a gamepad
Not mapping Logitech USB Receiver System Control as a gamepad
Not mapping rk805 pwrkey as a gamepad
Not mapping rockchip-hdmi0 rockchip-hdmi0 as a gamepad
Not mapping Google Pixel USB-C earbuds as a gamepad
Not mapping BRLTTY 6.3 Linux Screen Driver Keyboard as a gamepad
Stream 1280 x 720, 60 fps, 10000 kbps
RTSP port: 48010
Initializing platform...done
Resolving host name...done
Initializing audio stream...done
Starting RTSP handshake...Reference frame invalidation is not supported by this host
Audio port: 48000
Video port: 47998
Control port: 47999
done
Initializing control stream...done
Initializing video stream...done
Initializing input stream...done
Starting control stream...done
moonlight: /home/orangepi/moonlight-embedded/src/video/rk.c:507: rk_setup: Assertion `plane_id' failed.
Aborted

**What is the expected result?**
Stream should begin

**What happens instead of that?**
I get the error in the verbose about plane_id failed. I get the same error if I try -app "Steam Big Picture" which I have enabled on the host PC. When I do a moonlight list, Steam Big Picture and Desktop are the 2 options I get.

cgutman, 2023-09-02:
Are you running Moonlight from a separate TTY without Wayland or X running? Moonlight must have DRM master to output video.

royka1, 2023-09-08:
I'm getting exact the same error with the rk3588, ran from console and stopped the GDM service. I've seen that error message months ago too but then switched over to moonlight-qt.

MarkTwain007, 2023-09-08:
> I'm getting exact the same error with the rk3588, ran from console and stopped the GDM service. I've seen that error message months ago too but then switched over to moonlight-qt.

Were you able to get hardware decoding/ h.264 support working on moonlight-qt on the rk3588? I could not get it to work on the rk3588S

royka1, 2023-09-08:
Yes I've just rebuild it, works good but somehow I'm missing the icons. You'll need to have ffmpeg installed with rkmpp support.

[image]

kalmanchapman, 2023-10-01:
> Are you running Moonlight from a separate TTY without Wayland or X running? Moonlight Embedded must have DRM master to output video.

I'm having the same issue after building from source on an orangepi 5b running https://github.com/Joshua-Riek/ubuntu-rockchip

I'm able to run using `--platform sdl` but not with `--platform rk` which gives the same error as above.

cgutman, 2024-02-18:
This was probably fixed by #882

closed 2024-02-18 by cgutman