moonlight-stream/moonlight-embedded #877: Failure to fully compile on Raspberry Pi 3 with bookworm: interface/mmal/mmal.h: No such file or directory

doqfgc, 2023-10-29:
Please provide the following info.

**_NVidia Geforce Experience version:_** not applicable
**Moonlight Embedded version:** v2.6.1
**Moonlight Embedded source:** source
**Moonlight Embedded running on:** Raspberry Pi 3 (non-plus)
**Moonlight Embedded running on distribution:** DietPi (based on Raspberry Pi OS bookworm)

**Verbose output `-verbose` of Moonlight Embedded:** not applicable (doesn't compile)

**What is the expected result?**
Moonlight-Embedded should compile without issue

**What happens instead of that?**
truncated output
```
[ 72%] Linking C shared library libgamestream.so
[ 72%] Built target gamestream
[ 74%] Building C object CMakeFiles/moonlight.dir/src/config.c.o
[ 75%] Building C object CMakeFiles/moonlight.dir/src/connection.c.o
[ 77%] Building C object CMakeFiles/moonlight.dir/src/loop.c.o
[ 79%] Building C object CMakeFiles/moonlight.dir/src/main.c.o
[ 81%] Building C object CMakeFiles/moonlight.dir/src/platform.c.o
[ 82%] Building C object CMakeFiles/moonlight.dir/src/sdl.c.o
[ 84%] Building C object CMakeFiles/moonlight.dir/src/util.c.o
[ 86%] Building C object CMakeFiles/moonlight.dir/src/input/evdev.c.o
[ 87%] Building C object CMakeFiles/moonlight.dir/src/input/mapping.c.o
[ 89%] Building C object CMakeFiles/moonlight.dir/src/input/udev.c.o
[ 91%] Building C object CMakeFiles/moonlight.dir/src/audio/alsa.c.o
[ 93%] Building C object CMakeFiles/moonlight.dir/src/audio/pulse.c.o
[ 94%] Linking C executable moonlight
[ 94%] Built target moonlight
[ 96%] Building C object CMakeFiles/moonlight-mmal.dir/src/video/mmal.c.o
/root/moonlight-embedded/src/video/mmal.c:44:10: fatal error: interface/mmal/mmal.h: No such file or directory
   44 | #include <interface/mmal/mmal.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/moonlight-mmal.dir/build.make:76: CMakeFiles/moonlight-mmal.dir/src/video/mmal.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:150: CMakeFiles/moonlight-mmal.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

From what I've researched, this may not be included at all in bookworm-- https://github.com/raspberrypi/userland
> Code from here is no longer installed on latest RPiOS Bookworm images.

cgutman, 2023-10-29:
Apparently enough MMAL is still installed to fool our [MMAL detection](https://github.com/moonlight-stream/moonlight-embedded/blob/master/cmake/FindMMAL.cmake). Can you post the output of `cmake` itself and attach the `CMake*.log` files?

I don't know what the default KMS driver is on Bookworm on the Pi 3, but if they're using Full KMS by default, you'll need to switch to Fake KMS instead to use Moonlight Embedded. MMAL is incompatible with Full KMS, but the Raspberry Pi folks didn't replace it with a performance-equivalent alternate (especially on earlier Pis). The Full KMS compatible alternative is v4l2m2m and DRM, but that adds an additional frame of latency and can only barely sustain 1080p60 even on the Pi 4.  See this thread: https://github.com/jc-kynesim/rpi-ffmpeg/pull/25

If you want to build Moonlight Embedded for Pi OS Bookworm, you will need to manually compile and install https://github.com/raspberrypi/userland since Pi OS is no longer packaging this code. You can use my packaging scripts as a reference for doing this: https://github.com/cgutman/moonlight-embedded-packaging/blob/master/scripts/build-rpi-userland.sh

doqfgc, 2023-10-30:
> Can you post the output of `cmake` itself and attach the `CMake*.log` files?

Attached. Only one .log was generated.
[CMakeOutput.log](https://github.com/moonlight-stream/moonlight-embedded/files/13199687/CMakeOutput.log)

> I don't know what the default KMS driver is on Bookworm on the Pi 3, but if they're using Full KMS by default, you'll need to switch to Fake KMS instead to use Moonlight Embedded.

It looks like not only is Full KMS the default, it's the only option, at least on my system-- I didn't find any flags to switch anywhere.

It seems a bit excessive to have to restore deliberately removed code, but if there's no other means perhaps recomp from source is a bad idea. I did notice that `bookworm` packages dropped a few hours from this post so I'll likely just use those.

Also got a `error: size of array ‘vcos_static_assert’ is negative` building rpi-userland, which is a whole 'nother bag of chips.

cgutman, 2023-11-03:
The Bookworm packages have shipped in v2.6.2 and the wiki is updated with new steps required for installation on Bookworm.

closed 2023-11-03 by cgutman