moonlight-stream/moonlight-embedded #885: Problem trying to build moonlight embedded on RPI 3

BillBallong, 2024-02-12:
Please provide the following info.

**_NVidia Geforce Experience version:_**
**Moonlight Embedded version:** Moonlight Embedded v2.6.2
**Moonlight Embedded source:** https://github.com/moonlight-stream/moonlight-embedded/wiki/Full-details-for-Compilation-and-Installation-in-Raspbian-Or-OSMC
**Moonlight Embedded running on:** Raspberry Pi 3 model B rev 1.2
**Moonlight Embedded running on distribution:** Raspbian/Debian bullseye

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?** 
Trying to run cmake../ but it says it's not finding the requires packages.

**What happens instead of that?** 
~/moonlight-embedded/build $ cmake ../
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.30.2")
-- Found ALSA: /usr/lib/aarch64-linux-gnu/libasound.so (found version "1.2.4")
-- Found Opus: /usr/lib/aarch64-linux-gnu/libopus.so
-- Could NOT find Broadcom-OMX (missing: OPENMAXIL_LIBRARY)
-- Could NOT find Freescale (missing: FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find Amlogic (missing: AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY)
-- Could NOT find Rockchip (missing: ROCKCHIP_INCLUDE_DIR ROCKCHIP_LIBRARY)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2")
-- Checking for module 'libevdev'
--   Found libevdev, version 1.11.0
-- Checking for module 'libudev'
--   Found libudev, version 247
-- Checking for module 'sdl2>=2.0.4'
--   No package 'sdl2' found
-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
-- Checking for module 'libavutil'
--   No package 'libavutil' found
-- Checking for module 'vdpau'
--   No package 'vdpau' found
-- Checking for module 'libva'
--   No package 'libva' found
-- Checking for module 'egl'
--   No package 'egl' found
-- Checking for module 'glesv2'
--   No package 'glesv2' found
-- Checking for module 'x11'
--   No package 'x11' found
-- Checking for module 'libva-x11'
--   No package 'libva-x11' found
-- Checking for module 'libpulse-simple'
--   Found libpulse-simple, version 14.2
-- Checking for module 'libcec>=4'
--   No package 'libcec' found
-- Checking for module 'mmal'
--   Found mmal, version 1
CMake Warning (dev) at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (LibUUID).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindPkgConfig.cmake:59 (find_package_handle_standard_args)
  cmake/FindLibUUID.cmake:15 (INCLUDE)
  libgamestream/CMakeLists.txt:3 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for one of the modules 'uuid'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CURL: /usr/lib/aarch64-linux-gnu/libcurl.so (found version "7.74.0")
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found suitable version "1.1.1w", minimum required is "1.0.2")
-- Found EXPAT: /usr/lib/aarch64-linux-gnu/libexpat.so (found version "2.2.10")
-- Checking for module 'avahi-client'
--   Found avahi-client, version 0.8
-- Performing Test HAVE_GETAUXVAL
-- Performing Test HAVE_GETAUXVAL - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rpi/moonlight-embedded/build

Legincy, 2024-02-17:
Same problem, any news on that?

cgutman, 2024-02-17:
There is no error in your provided logs. It says it successfully generated build files.

BillBallong, 2024-02-17:
That's good then, I was thinking there was a problem with not finding the correct packages to build the files because when I run 

"make -j$(nproc)" and "sudo make install"

There are 2 errors including missing some files.

rpi@raspberrypi:~/moonlight-embedded/build $ make -j$(nproc)
[  1%] Built target docs
[ 56%] Built target moonlight-common
[ 74%] Built target gamestream
[ 75%] Building C object CMakeFiles/moonlight-mmal.dir/src/video/mmal.c.o
[ 98%] Built target moonlight
/home/rpi/moonlight-embedded/src/video/mmal.c:44:10: fatal error: interface/mmal                                   /mmal.h: No such file or directory
   44 | #include <interface/mmal/mmal.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/moonlight-mmal.dir/build.make:82: CMakeFiles/moonlight-                                   mmal.dir/src/video/mmal.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/moonlight-mmal.dir/all] Error                                    2
make: *** [Makefile:149: all] Error 2

rpi@raspberrypi:~/moonlight-embedded/build $ sudo make install
[ 55%] Built target moonlight-common
[ 72%] Built target gamestream
[ 74%] Building C object CMakeFiles/moonlight-mmal.dir/src/video/mmal.c.o
/home/rpi/moonlight-embedded/src/video/mmal.c:44:10: fatal error: interface/mmal/mmal.h: No such file or directory
   44 | #include <interface/mmal/mmal.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/moonlight-mmal.dir/build.make:82: CMakeFiles/moonlight-mmal.dir/src/video/mmal.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/moonlight-mmal.dir/all] Error 2
make: *** [Makefile:149: all] Error 2

cgutman, 2024-02-17:
Are you running the 64-bit version of Raspberry Pi OS? I think they only officially support MMAL with the 32-bit version (though it _does_ work for what we use it for). You can try to install `libraspberrypi-dev` but if it doesn't fix the issue, you will have to build the libraries yourself. 

I would recommend just sticking to the official `moonlight-embedded` packages, which handle all this for you.

But if you really want to compile yourself, you will need to use the `static_libs` branch of [my fork](https://github.com/cgutman/userland/tree/static_libs) of the Raspberry Pi userland repository, then compile and install it according to the [packaging script](https://github.com/cgutman/moonlight-embedded-packaging/blob/master/scripts/build-rpi-userland.sh), and finally try to build again.

closed 2026-10-26 by BillBallong