moonlight-stream/moonlight-embedded #886: Add option to flip face buttons

denizonm, 2024-02-16:
Add option to make the buttons a<->b x<->y, please.

Sunshine Version: 0.21.0.5bca024.dirty
Moonlight Embedded version: 2.6.2
Moonlight Embedded source: Comes with jelos
Moonlight Embedded running on: RGB30
Moonlight Embedded running on distribution: JELOS

for some reason the gamepad mapping flip the face buttons by default, the only way to fix being an edited gamecontrollerdb.txt (i could only get this fix working on arkos and not jelos but i am probably missing something)

cgutman, 2024-02-20:
Which gamepad?

pettersondiniz, 2026-02-15:
I tried to flip these buttons all the day and I failed

I am using a RG35XXH and the face buttons are in Nintendo pattern. So, when I connect to host, the position are wrong. I tried to give a custom mapping and nothing works

Moonlight Embedded 2.7.0-master-274d3db

OS: MustardOS 2508.4 (LOOSE_GOOSE) aarch64                         ⠀⠀⠀⠀
Kernel: Linux 4.9.170 

Apollo 0.4.6

pettersondiniz, 2026-02-15:
Reading the files and the scripts I finaly discover how to swap the buttons. 
It's a global config, not only for moonlight. 

In a script I've found a command that refered a config file that reads a value from /opt/config/settings/advanced/swap
the value can be 0 or 1.  based on the value of this parameter, the function loads 'modern.txt' or 'retro.txt'.

aftre that I followed the path on config menu > general settings > advanced settings > button swap. I put it in modern and worked for me, but is affected all the system.